MatterHackers/PulseV1Firmware #31: Bring Pulse in line with what is advertised on the web site.

martinpirringer, 2018-12-27:
This might fall under wish list. But it irks me. I would like to perform the pulse to at least what it says on the web site. the most egregious short coming is the advertised E3D V6 upgrade. where it states on the web site and was repeated by your sales staff

Max. Temp | E3D Lite6 245°CE3D V6 (Upgrade) >400°C.

Now I would like it to do that. I have the E3D V6 upgrade BUT according to E3D the V6 on the Pulse is in no shape to get anywhere near that. First it has the standard thermistor which is rated to top out at 295 and will take damage if run for extended periods of time above 285. Now E3D will sell the PT100 which goes to 400 (for >400 you need a thermocouple) Now I tried to get the PT100 but was told the pulse firmware cannot do it and there are no plans of doing it (according to E3D all that would be needed is to change the probe type from 5 to 20 and properly wire the probe and amplifier board. Now with that you still cannot got to 400 though as the aluminum block that comes with the E3D V6 upgrade will degrade at 350. To go above 350 you need the copper block. Further more at that point the plastic pieces most likely will have to be replaced and most likely the heater cartridge to the higher powered one. Somewhere around 320 the Silicon socks will degrade so E3D suggests to run it without socks if you go to the copper block and the PT100. Now on the C232 that I bought the plastic parts at the hotend are made out of Ryno which gets soft at about 70 C so I doubt that it will survive a 400 C uncovered heater block about 1/8 in or so away. So the hotend parts would have to be printed out of Nylon or some other higher temp material or made out of metal. 

2nd smaller issue is the bed. Besides the wiring which should have a strain relief both top and bottom to avoid yanking on big parts the bed should either be replaced with a higher powered one or insulated on the bottom. According to the specs the Pulse can print ABS. Yes it can but if you go print something big ABS that takes the whole build plate the temp needs to get cranked up and held at 100 or even higher if you want to be successful with MH build ABS on a garolite bed but it takes a ton of effort and getting past multiple "failure to heat" errors to make it to 100 even with an enclosure 

Printer Pulse C232.
-- | --

larsbrubaker, 2018-12-28:
@cope413, can you have a look at this? Thanks.

RSteel517, 2019-01-07:
To add onto this, the Pulse is most similar to the Prusa i3 Mk2 (ish).  With that in mind, being able to perform a mesh bed level and utilize the filament run out sensor WITHOUT requiring connection to MatterControl would be nice (similar to the Pulse's competitors).

refriedchicken, 2020-04-23:
@RSteel517 you can add G29 to start gcode and set the BLtouch in the firmware with M851. I am using one of my Pulses with Octprint and PrusaSlicer with this method.