MatterHackers/PulseV1Firmware #47: Cant compile

ibbr0, 2021-04-26:
I can't compile the eframe project using vscode with platformio and autobuildmarlin. I get the following error

Compiling .pio\build\rambo\src\src\sd\SdFatUtil.cpp.o
In file included from Marlin\src\module\../inc/MarlinConfig.h:47:0,
                 from Marlin\src\module\endstops.h:28,
                 from Marlin\src\module\endstops.cpp:27:
Marlin\src\module\endstops.cpp: In static member function 'static void Endstops::report_states()':
Marlin\src\module\endstops.cpp:492:46: error: 'xyze_long_t Stepper::count_position' is private within this context
    SERIAL_ECHOLNPAIR_F("stepper e=", stepper.count_position.e);
                                              ^
I can compile the release from marlinfirmware with no problem, but there are hundreds of differences.

beandropcompany, 2021-05-23:
I'm having the same issue with compiling. Any help or guidance will be apprieciated

ibbr0, 2021-05-23:
The only solution I found was to comment out the 'SERIAL_ECHOLNPAIR_F("stepper e=", stepper.count_position.e);'. That allows it to compile and I was able to run the pulse with it. I got some general direction about rolling back libraries from MH tech support with no specifics. And as you can see above the MH guy whose name is on the repo took the time to change 'Cant' to 'Can't' in my orginal post. (Thanks) However I did see some 'driver overheat, current reduced' messages. I managed to find a zip from a newer non supported repo of the actual MH pulse code which compiles immediately after specifying the proper board and I've been using that. I could share or email a copy if you have a way of getting 7.1mb. I found in that code, everytime I tried to implement relatively minor changes, it would throw errors. For instance disabling filament sensor made it not compile. Parking the head required to support separate filament load/unload commands gave crazy results when using ubl bed leveling. Not knowing how much of this is from MH 2.0.6 based repo and how much is Marlin itself, Yesterday I switched to 2.0.8.1 from the Marlin repo. (not MH). Just got it running last night and successfully probed the bed (from MC) and printed a part. I had to tweak the EinsyRambo pin file just a bit to get the BLTouch probe working but everything else was in the 2 config and config_adv. There are many features not turned on yet, Leveling, filament change/load/unload, Linear advance but it is straight from Marlin and I hope easier to get community support. (I'm a noob at Marlin, but I know how I want to run my pulse). I am willling to share the 3 files changed, or a zip of the project. I will not guarantee it or support it other than in this or other forums, but it is a working starting point if yo know what you are doing.

beandropcompany, 2021-05-24:
Thank's for the guidance. I've been trying to convert the Pulse from a Bowden setup to a direct drive, I used a Thingiverse print (https://www.thingiverse.com/thing:3510052) to attach the Bondtech extruder directly above and reversed the wiring on the stepper motor. I wanted to detach the printer from Matter Control and as you have and take the firmware straight from marlin. I had the same problems making small changes to the firmware before trying to compile such as swapping out the BLTouch for a PINDA v2 probe. (I'm basically attempting to convert the Pulse into a Prusa to improve quality and reliability.  But due to axis size differences, just flashing the Prusa Firmware won't work as it fails on calibration before you're even able to print, and that's a whole other mess of tweaks to bypass). In the end, I reverted back to MatterControl last night, temporarily fixed the BLTouch to the side of the new direct drive set up with electrical tape (It's a hack), and amazingly it worked with changes to retraction settings in relation to the direct drive.  I'm going to print a new direct-drive bracket to fit the BLTouch permanently and go from there. I'm not sure I can quite face attempting to detach the printer from Matter Control again for another few weeks as I'm desperate for some prints for the business and the printer has been offline for one reason or another for the last 3 weeks. If you are able to drop the files here(https://drive.google.com/drive/folders/1gDW8zvar6IBZVmOx3kx2TY6ZNHDDioSl?usp=sharing), I will face the music and play around with them in a few weeks. If not, no worries, I will take comfort knowing someone else has had partial success and that it is possible. Thank you!

ibbr0, 2021-05-24:
Files should be there Sent from Mail for Windows 10 From: Bean DropSent: Monday, May 24, 2021 3:08 AMTo: MatterHackers/PulseV1FirmwareCc: ibbr0; AuthorSubject: Re: [MatterHackers/PulseV1Firmware] Can't compile (#47) Thank's for the guidance. I've been trying to convert the Pulse from a Bowden setup to a direct drive, I used a Thingiverse print (https://www.thingiverse.com/thing:3510052) to attach the Bondtech extruder directly above and reversed the wiring on the stepper motor. I wanted to detach the printer from Matter Control and as you have and take the firmware straight from marlin. I had the same problems making small changes to the firmware before trying to compile such as swapping out the BLTouch for a PINDA v2 probe. (I'm basically attempting to convert the Pulse into a Prusa to improve quality and reliability. But due to axis size differences, just flashing the Prusa Firmware won't work as it fails on calibration before you're even able to print, and that's a whole other mess of tweaks to bypass). In the end, I reverted back to MatterControl last night, temporarily fixed the BLTouch to the side of the new direct drive set up with electrical tape (It's a hack), and amazingly it worked with changes to retraction settings in relation to the direct drive. I'm going to print a new direct-drive bracket to fit the BLTouch permanently and go from there. I'm not sure I can quite face attempting to detach the printer from Matter Control again for another few weeks as I'm desperate for some prints for the business and the printer has been offline for one reason or another for the last 3 weeks. If you are able to drop the files here(https://drive.google.com/drive/folders/1gDW8zvar6IBZVmOx3kx2TY6ZNHDDioSl?usp=sharing), I will face the music and play around with them in a few weeks. If not, no worries, I will take comfort knowing someone else has had partial success and that it is possible. Thank you!

beandropcompany, 2021-05-27:
Thank you very much! I will feedback when I move over to it in a few weeks.